monome/teletype #11: MIDI

tehn, 2016-03-28:
it's very easy to get MIDI input via USB. of course, this means not having a keyboard plugged in, so much swapping would be required. (the alternative approach here is an tt-extension where this functionality is over II)

three new "events" (ie metro, trigger, etc): NOTE-ON, NOTE-OFF, and CC

with a note-on event, a pre-determined variable NOTE (or whatever) is set, and then this script does whatever with that information. same with off, and with cc.

basically just associating scripts with incoming midi events. and then TT becomes a crazy flexible midi processor.

burnsauce, 2017-09-03:
Could the TT support a USB hub?

samdoshi, 2017-09-03:
Not easily 😢 

Someone would have to code support for it into `libavr32` or the Atmel Software Framework. There are a bunch of threads about it on lines.

tehn, 2017-10-12:
closing this foggy idea until someone reimagines it.

closed 2017-10-12 by tehn