monome/teletype #14: comment line

tehn, 2016-03-28:
hot key to comment out a single line.

indicated with brightness change.

etherbrooklyn, 2017-01-07:
Is this to allow, say, // comments in scripts?  That would be ignored when loading from USB?

I'd appreciate that, as I've been working on compiling all the scrips from various lllllll threads into one nice collection, and would like to add inline comments.

genericwoods, 2017-01-09:
I believe this would be for debugging scripts when working in the Teletype itself. You could easily 'turn off' a line of code, without deleting it. At least that’s how interpreted it. @etherbrooklyn your suggestion also sounds useful in a different way, you may want to add another feature request for that.

burnsauce, 2017-09-04:
The patch submitted has comments persist through save/load FYI.  If this is not desirable, they can be scrubbed before saving.

jlmitch5, 2017-10-09:
@burnsauce not something that would be necessary to change in the 2.1 timeframe by any means, but my thinking is not-persisting on save would possibly be a little "safer" in terms of UX.

Consider if the user loads a script that has a single commented-out line.  Because our indication of commenting is making the line dim, it could lead to it being unclear as to _why_ the line isn't responding to a trigger.

cmcavoy, 2017-10-10:
> @burnsauce not something that would be necessary to change in the 2.1 timeframe by any means, but my thinking is not-persisting on save would possibly be a little "safer" in terms of UX.

Doesn't this solve #104 as a by-product? Allowing comments without changing the serialization syntax (which would force a major version change on release). If we kept the `//comments` in the serialized output, we make it easier for people to post comments to their uploaded scripts on https://github.com/monome-community/teletype-codex

cmcavoy, 2017-10-11:
I had a conversation with one of the key folks working on Teletype Codex, #104  isn't a blocker like I thought. This is a cool feature though, 👍

commit ad358cdb (2017-10-12): 2.1.0 Release (#114)

* Fixes #35 by guarding SCENE from initial I script

Added an initialization flag to the scene state to denote when a scene
is first loading.  SCENE operator aborts if the flag is present.

* Fixes #63 by overhauling data entry in tracker

Buffered input
Clear-on-new-entry
Apply-on-Enter
Abort-on-navigation
Incrementers wrap around 16-bit limits
Number entry blocks if it would exceed limits

* New Op: MSPB (milliseconds per beat)

* Fixes #14 by allowing code commenting with alt-/

* New Op: LAST - delta between script run times

Now tracking script number in the executio [... cut]
commit e33acf1d (2017-10-18): Fixes #14 by allowing code commenting with alt-/
closed 2017-10-12 by tehn