monome/teletype #7: tracker alternate data views

tehn, 2016-03-28:
hot key to toggle view mode:

* normal (numbers)
* bars, coarse (voltage)
* bars, fine (notes)
* bars, fine, bipolar (notes relative)
* boxes? (on/off)

how to deal with range in fine bars mode?

burnsauce, 2017-09-03:
Sounds doable.

Range in fine bars mode could change with another hotkey through 1, 2, 4, and 8 octaves.  How to signify what mode is being displayed, though?  If there's not the screen real estate to have it indicated all the time, there could be a brief pop-up box when it changes, and the same could be said for cycling the tracker view mode.

tehn, 2017-09-12:
i feel this needs some substantial design before we dig in-- give me some time to work on it, or i'm open to design propositions.

differentiation should be visual. ie, a VOLT bar and NOTE bar should look different, a bipolar bar should have a tick in the center, etc. so there shouldn't need to be indication.

closed 2023-06-16 by tehn