monome/teletype #8: TL (timeline)

tehn, 2016-03-28:
a linear script system, sortof like csound scores. which i admit, i have no experience with, so please chime in with use cases that have already been addressed and solved that i'm overlooking.

this functionality would add a linear "score" of commands, that can be non-linearly executed with other teletype commands.

this would entail making another editor screen, like METRO, but may need more interactivity. each line would have an index or time marker. sequential lines can have the same time. ie:

```
1 TR.PULSE A
1 TR.PULSE B
2 TR.PULSE A
4 TR.PULSE A
10 TR.PULSE A
```

of course add more interesting commands. i hope this buffer can be very long with many many events.

`TL` gets/sets the timeline position.
`TL.NOW` executes the current position.
`TL.NEXT` adds one to the position and executes.
`TL.START` sets the start position (for control by `TL.NEXT`)
`TL.END` sets the end position

* there can be unpopulated index positions, ie instances where `TL.NOW` will do nothing.
* i like the idea of there not being a hard limit to how many commands can be on one index.
* the editor should facilitate "insert" commands where later indexes will be incremented and smart "delete" where following indexes are decremented, etc

tehn, 2016-03-28:
comments by @samdoshi 

The timeline stuff seems really really cool but maybe not just as a timeline...

Would you be able to add direct execution of a time? E.g. TR.EXEC t, because now I can treat it as a function table... maybe something along the lines of:

TIMELINE

0 X ADD X 5
5 X SUB X 3
10 X 0
SCRIPT 1

TR.EXEC RAND 10
CV A N X
As an aside, I'd probably add an OP in a private branch for my own use to treat the time as a probability. Maybe something like TR.EXECF t (execute the first time after t), then in the example above you could guarantee that one of the instructions runs... I've been thinking about Markov chains recently...

The other slightly out-there idea, would be to allow code to change the times, e.g. TL.MAP t1 t2 changes all the commands at t1 to t2. Also maybe TL.SWAP t1 t2

tehn, 2017-09-06:
revised spec:

### new UI screen, TL editor

* scrolling. need indication for scroll position, hot keys, etc.
* methods for moving lines around. copy/paste, shifting up/down

### commands

`TL.CALL` - execute line (x)
`TL` - execute line at position
`TL.POS x` - set position (also get with no arg)
`TL.NEXT` - set next position (respect bounds)
`TL.PREV` - set previous position (respect bounds)
`TL.START x` - set start (also get with no arg)
`TL.END x` - set end (also get with no arg)
`TL.SEEK` - move position forward to next command (populated line)

possible commands for editing the TL data itself

`TL.MOVE x y` - move lines at index x to index y
`TL.SWAP x y` - swap lines at index x with lines at index y
`TL.CLEAR x` - clear lines at index x

considering commands that change position and also execute, though perhaps this is sufficient:

`TL.NEXT; TL`

samdoshi, 2017-09-07:
Timeline commands are supposed to be saved with the scene right? In which case attention will need to be given to the USB disk code too.

Another OP... (or rather MOD):

`TL.SET x: <cmd>` - save `<cmd>` to position `x`

closed 2020-07-16 by tehn