monome/teletype #9: AT (autotrig)

tehn, 2016-03-28:
there have been requests for more metro scripts, and i think this is a good idea. however, tracking them and making more screens would clutter the interface. this is one solution.

each trigger input 1-8 would have an additional "autotrigger" functionality. 

`AT.ACT x y` would activate/deactivate autotrig on channel input x (y=1 on, y=0 off)
`AT.TIME x y` sets the interval time of channel x to y
`AT.COUNT x y` sets how many repeats (0 = infinite repeat)
`AT x` would trigger channel x (this seems redundant with `SCRIPT`)

so. for example, activate autotrig on channel 1. set time to 500 (ms). set count to 0. go `AT 1` and suddenly you have another metro that executes script 1 at its own rate.

another use: script echo. set count to 4. remote trigger inputs (via patch cable) will also trigger the AT. retrigger resets count.

external trigger also resets the per-trigger metro counter (as does another AT command).

tehn, 2016-03-28:
comments by chrisharrisx

I agree that AT x seems somewhat redundant in light of SCRIPT, so what about the idea of using AT as a single line setter/activator, as in
AT 1 500 0 – enable autotrig on channel 1 with 500ms and infinite repeats?

I can certainly still see using ACT, TIME, and COUNT as individual commands to set or alter a running autotrig, but in any cases where you want less than infinite repeats AT would be a real space-saver. Not only that, but it should theoretically also enable us to combine autotrigger with PREs, no?

I'm imagining something like:
L 1 4 : AT I DIV 500 I 4
or
IF a : AT 1 500 0

burnsauce, 2017-09-01:
With the cut-down syntax, what would be an appropriate idiom for disabling the AT?
Set COUNT to 1? i.e.: AT 1 1 1? (No special implementation required)
Set TIME to 0? i.e.: AT 1 0 X? (1 conditional)
Special syntax? e.g.: AT 1 0 (parser case due to fewer arguments), AT 1 (like a toggle, same: parser)?

I kind of like the toggle concept, but it requires us to select sane defaults for TIME so that an init-fresh call of AT 1 will make sense.  Maybe 500 or 250?

tehn, 2017-09-06:
disabling with AT.ACT x 0

but AT and SCRIPT are identical, so let's have AT.ACT be just AT.

so:

AT x y where x is the script (1-8) and y is on/off (1/0)

AT x would simply return the on/off state of that script

for toggle, AT (x) EZ AT (x) where x is the script to toggle

default AT.TIME to 500, COUNT to 0 (infinite)

yes? i think this will be a massively powerful addition!

burnsauce, 2017-09-06:
Looks good.  I'll put it on the list.

trentgill, 2017-09-09:
This idea seems great, however I'd like to critique the name, and chosen abbreviation AT. To me this reads as at, as in @ or 'located at'.

I considered ATR though 'TR' in teletype means the TR outputs, not the 'SCRIPT' input jacks.

Also something more plain-english like forth's AGAIN, though this more specifically refers to recursion and wouldn't make sense when working across different scripts.

tehn, 2017-09-09:
will take this to the forum, good point

burnsauce, 2017-09-12:
I think forum favours R for Repeat.

tehn, 2017-09-12:
R is is.

closed 2020-07-16 by tehn