monome/teletype #28: Setup travis build.

pq, 2016-04-06:
A few bits:

- [ ] setup `.travis.ym;` appropriately --- @samdoshi has proved this out (https://github.com/samdoshi/teletype/commit/0a51dd8c1efd6fc3865cc624b80d52108b2ec5af), so this shouldn't be too hard
- [ ] add a build badge to the `README`

pq, 2016-04-06:
@tehn : looks like an admin needs to flip the bit on travis for teletype: 

[image]

https://travis-ci.org/profile/monome

Could you?

tehn, 2016-04-06:
flipped

tehn, 2016-04-06:
i must say, i am completely baffled by the contents of .travis.yml

pq, 2016-04-06:
Ha!  I blame @samdoshi.

tehn, 2016-04-06:
ah, it makes more sense now that i actually read about travis.

so right now we're just doing a compilation check, correct?

pq, 2016-04-06:
> so right now we're just doing a compilation check, correct?

Right. Once we have tests we should add them.  Down the road we could add linting or some other kind of presubmit checks.  Note that pull requests trigger a build which means you can see if a PR will breaj the build before accepting it.

For example:

[image]

pq, 2016-04-06:
Added badge with https://github.com/monome/teletype/commit/cbb9fe8eb7613dfa91ccdc9d16a9953af4165b02 which means this first bit is done.

@samdoshi or @tehn : feel free to re-open or create another issue if you want to see refinements to the travis build.

closed 2016-04-06 by pq

tehn, 2016-04-06:
looks good, thanks!

samdoshi, 2016-04-07:
> i must say, i am completely baffled by the contents of .travis.yml

You and me both.

Should we get https://github.com/samdoshi/avr32-toolchain-linux move over too?

pq, 2016-04-07:
> Should we get https://github.com/samdoshi/avr32-toolchain-linux moved over to the 'monome' org too?

:+1: 

Make sense to me!