monome/teletype #34: Infinite loop with SCRIPT

samdoshi, 2016-04-10:
There is some protection against causing an infinite recursion loop with the `SCRIPT` op, but it's easily worked around, e.g.

1:
```
SCRIPT 2
```

2:
```
SCRIPT 3
```

3:
```
SCRIPT 1
```

If you trigger script 1, you will lock up the device. (Worse if someone 

The problem will only be worse with timelines, so we need to rethink how we deal with this.

One alternative could be to allow a maximum number of commands to be executed by an external event (e.g. trigger, metronome, delayed commands), and then not worry about the infinite loop.

tehn, 2016-04-11:
ahh. the `script_caller` set/reset needs to be repaired, you are correct.

`
int8_t script_caller;
static void tele_script(uint8_t a) {
    if (!script_caller) {
        script_caller = a;
        for (int i = 0; i < script[a - 1].l; i++) process(&script[a - 1].c[i]);
    }
    else if (a != script_caller) {
        for (int i = 0; i < script[a - 1].l; i++) process(&script[a - 1].c[i]);
    }

    script_caller = 0;
}
`

tehn, 2016-04-11:
the script_caller var should be set to zero only by the initiating call to tele_script.

samdoshi, 2016-04-12:
I think you'll still be able trigger infinite recursion so long as you don't visit the initiation script, e.g.

1:
```
SCRIPT 2
```

2:
```
SCRIPT 3
```

3:
```
SCRIPT 2
```

If you want to disallow any form of recursion, we need to track the call stack, at which point we can decide if we want to allow no recursion, or a limited amount.

If we go with no recursion we still need to stop running commands on call stack overflow so that we fail safe.

I'm planning on introducing the concept of an 'execution' context which can hold things like the call stack.

samdoshi, 2017-05-17:
Fixed in #73

closed 2017-05-17 by samdoshi