monome/teletype #36: Should variables be initialised on SCENE load?

samdoshi, 2016-04-12:
At the moment variables stay at the previously set value when switching scenes. Should they?

You don't have enough lines in the `I` script to initialise all of them.

samdoshi, 2016-04-12:
The other option is to save the current value for all variables in the preset

tehn, 2016-04-12:
saving the variable state in the scene seems like it introduces some
obfuscation as to how a scene works. i'd rather have things be set in INIT.

but yes, there's only 6 lines (can be more by using dummy scripts or
eventually timeline).

perhaps we should introduce `CLEAR` which would zero out runtime variables.

note that the pattern data is considered separate and is indeed stored with
scenes.

burnsauce, 2017-09-03:
How about a Variables screen that can both serve as a live view and a method to initialize variables?

My init scripts are crowded enough as it is, and live view was already on my radar.

tehn, 2017-09-06:
TL (timeline) is a possible solution for this.

designing value-preset systems is tricky business-- i wouldn't suggest we save them in scenes or use live view/etc.

closed 2017-09-12 by tehn