monome/teletype #37: Bug with O?

samdoshi, 2016-04-12:
On a freshly booted up Teletype...
```
O        prints 1
O 0
O        prints 1
```
The docs indicate that it should return a 0. Want me to fix it?

tehn, 2016-04-12:
certainly. perhaps the increment is happening before return?

samdoshi, 2016-04-12:
It is, I just wanted to check which was wrong, the docs or the implementation.

How do you feel about renaming `O.DIR` to `O.INC` to indicate that it can have values other that `-1` and `1`?

I'll update the changelog if I make the change.

tehn, 2016-04-12:
INC makes more sense, agreed.

i think post-increment is also more understandable.

samdoshi, 2016-04-12:
Also spotted that `FLIP` was using `V_O`:

```c
static void v_FLIP() {
    if (left || top == 0) {
        push(tele_vars[V_O].v);
        tele_vars[V_O].v = (tele_vars[V_O].v == 0);
    }
    else {
        tele_vars[V_O].v = (pop() != 0);
    }
}
```

tehn, 2016-04-12:
good catch!

samdoshi, 2016-04-12:
`FLIP` fixed
https://github.com/samdoshi/teletype/commit/048559fa91ec6a3f4f80cdad239373586880bc4e

samdoshi, 2016-04-12:
`O` fixed
https://github.com/samdoshi/teletype/commit/06b79a3923376af12dcd176b8d9d34961eba9fef

samdoshi, 2016-04-12:
`O.DIR` renamed to `O.INC`
https://github.com/samdoshi/teletype/commit/a757a0c7b497fab5a0e8e47675d4fbd2615e0557

That's everything on this ticket, so I'm closing it

closed 2016-04-12 by samdoshi

samdoshi, 2016-04-12:
Have also updated `DRUNK` to follow the same semantics (as per the docs)

> DRUNK changes by -1, 0, or 1 upon each read, saving its state. Setting DRUNK will give it a new value for the next read, and drunkedness will continue on from there with subsequent reads.

https://github.com/samdoshi/teletype/commit/f720fced47edc762f5b656e49ddf08d8d1835ded