monome/teletype #51: Remove OP docs and help screen

samdoshi, 2016-10-14:
Do we want to remove the `.doc` field from ops and mods? The data is currently unused.

Same thing goes for the help text under the `F1` key.

There are currently 3 copies of the teletype docs, the website, the PDFs, and the source code. I think that's too many to try and keep up to date.

I was thinking we could use the `F1` - `F10` keys as script triggers. Maybe with a modifier to take you to the edit page for the script.

scanner-darkly, 2016-10-14:
agreed `.doc` could be removed, i don't see it being used anywhere. not sure if @tehn had plans for it?

i do find the help screen useful though but i also like the idea of using `F` keys as script shortcuts (with potentially more modifiers added later, like `CTRL` to copy, `SHIFT` to paste), so perhaps a different button could be assigned to it?

re: docs - should we revisit the idea of annotating in the source code and then using something like Doxygen to generate whatever content is needed? i haven't used Doxygen and did a quick search for simpler alternatives, there is https://github.com/libgit2/docurium and http://jessevdk.github.io/cldoc/gettingstarted.html but doesn't look like they would work well in this case. any concerns with using Doxygen?

tehn, 2016-10-14:
> cldoc/gettingstarted.html but doesn't look like they would work well in

i'm good to remove the .doc

tehn, 2016-10-14:
script triggers already assigned to meta-# i think?

tehn, 2016-10-14:
(sorry for the multiple replies.)

jump-to-edit-page is a great idea.

samdoshi, 2016-10-15:
PR in for `.doc` removal.

@scanner-darkly what do you find useful on the `F1` screen? The keys reference, the ops reference, or is it both.

Regarding Doxygen, what's your anticipated use? Developer docs or user docs?

scanner-darkly, 2016-10-16:
mostly the ops. but it's just so handy to have a quick ref right there in the module itself. 

doxygen - i think the benefit would be to do both, otherwise it's just one more thing to maintain. if it's possible to configure it to output for multiple formats that would mean it only needs to be maintained in one place. i need to read up on it.

samdoshi, 2017-05-17:
Fixed in #73

burnsauce, 2017-09-03:
Fixed?

burnsauce, 2017-09-05:
help_mode.c still exists and its keystrokes are still documented in modes.md.  Is this functionality dead?  Can it be culled?

samdoshi, 2017-09-05:
> help_mode.c still exists and its keystrokes are still documented in modes.md. Is this functionality dead? Can it be culled?

Enough people found it useful that it was decided to keep it. It still works, but the key binding is `alt-?`

closed 2017-09-05 by samdoshi