monome/teletype #63: Tracker Data Entry Improvements

burnsauce, 2017-01-17:
The data entry on the tracker screen should be buffered, i.e.: users should have to press 'ENTER' to confirm data, and could hit 'ESC' to cancel entry.

It would permit users to enter large values into patterns instantaneously.  Currently, intermediary typed values are caught by scripts reading the patterns at high rates -- any rate higher than the data entry speed.

Additionally, this would eliminate the problem of needing to delete the existing data in the field before entering data.

burnsauce, 2017-09-03:
As ESC is reserved, the implementation put forth uses all non-data-entry, yet valid keystrokes to abort data entry.

My only gripe is that I selected _append-on-entry_ behaviour for nonzero pattern values instead of _clear-on-entry_, as I would have preferred.  I selected the append behaviour because it followed the existing data entry model and it might be muscle memory for some users.  It's an easy fix if we agree that clear-on-entry is the better model.

tehn, 2017-09-06:
clear on entry makes sense.

commit ad358cdb (2017-10-12): 2.1.0 Release (#114)

* Fixes #35 by guarding SCENE from initial I script

Added an initialization flag to the scene state to denote when a scene
is first loading.  SCENE operator aborts if the flag is present.

* Fixes #63 by overhauling data entry in tracker

Buffered input
Clear-on-new-entry
Apply-on-Enter
Abort-on-navigation
Incrementers wrap around 16-bit limits
Number entry blocks if it would exceed limits

* New Op: MSPB (milliseconds per beat)

* Fixes #14 by allowing code commenting with alt-/

* New Op: LAST - delta between script run times

Now tracking script number in the executio [... cut]
commit 9f212103 (2017-10-18): Fixes #63 by overhauling data entry in tracker

Buffered input
Clear-on-new-entry
Apply-on-Enter
Abort-on-navigation
Incrementers wrap around 16-bit limits
Number entry blocks if it would exceed limits
closed 2017-10-12 by tehn