monome/teletype #72: PREs are allowed without MODs

samdoshi, 2017-03-06:
As far as I can tell this bug has been there since the beginning. Or at least for quite a long time.

Consider the following

```
TR.PULSE 1 : TR.PULSE 2
```

That command doesn't make any sense as the `PRE` (i.e. before the `:`) has no `MOD` statement (e.g. `IF`). But it passes validation and runs!

What happens is the parser splits the entire command into a `PRE` and a `POST`, the validation step doesn't check to see if the first `WORD` of the `PRE` is a `MOD`. And finally the processor only executes the `PRE` (assuming that execution of the `POST` is delegated to the non-existent `MOD`).

The fix is fairly straightforward, we just need to insert a check at the end of `validate` to assert that if there is a `PRE` it must start with a `MOD`.

Unless I hear any objections I'll get it fixed in my parser branch.

tehn, 2017-03-06:
thank you for spotting this, and the fix sounds correct.

samdoshi, 2017-03-09:
Fixed here:

https://github.com/samdoshi/teletype/commit/e70ec4bf4eb11058f7c829706521c1e3979eb812

samdoshi, 2017-05-17:
Fixed in #73

closed 2017-05-17 by samdoshi