monome/teletype #79: planck keyboard compatibility

capogreco, 2017-07-25:
I would love to be able to use my planck to program the teletype, but the teletype doesn't even register its existence when I plug it in.

the planck uses Atmel's Atmega32u4 AVR controller, and the firmware, as well as other information, can be found [here](https://olkb.com/reference/primer/).

I would love to help on this one, but I might need some pointers as I am a very beginner level programmer, and I'm not sure where I would even start.  any ideas?

tehn, 2017-07-25:
does the plank display itself as a standard HID keyboard? if so it should work. if it requires any special drivers, it will not work.

samdoshi, 2017-07-25:
I've got a Planck keyboard (need to sell it really as I don't like it). I'll try and find some time in the next few days to test mine.

samdoshi, 2017-07-25:
Just tried my Planck and it works fine.

For your reference, it's got a REV2 PCB in it and the firmware build is probably from about 8 months ago.

capogreco, 2017-07-26:
uggg. sorry my bad - I just installed the new firmware and hadn't power cycled the teletype. it works!

samdoshi, 2017-07-26:
No worries @capogreco, happens to all of us...

closed 2017-07-26 by samdoshi