monome/teletype #81: USB Filesystem / Behaviour Improvements

burnsauce, 2017-09-03:
Bank (folder) swapping and selective backup (and/or nondestructive auto-backup) would go a long way toward a decent user experience.

A proper file browser wouldn't be a bad long-term goal.

samdoshi, 2017-09-03:
In my opinion, this file: 

https://github.com/monome/teletype/blob/master/module/usb_disk_mode.c

needs completely rewriting first. I have an outline in my head for how to do it, but it's going to be time consuming and not much fun (essentially split it into a "pure" function in `src` with tests and an effectful function in `module`), if you ever feel like doing it start a thread on the main forum and I will try to share my ideas.

(Although if you're doing the work, you're free to do it however you like....)

tehn, 2017-09-06:
a *minimal* file browser would certainly help, yes.

csboling, 2020-07-04:
Lots of discussion here, old thread at this point: https://llllllll.co/t/teletype-usb-disk-mode-interface/

closed 2020-07-16 by tehn