monome/teletype #102: Should @SCRIPT N be leaving a "return value" on the stack?

burnsauce, 2017-09-20:
If the final process_command leaves us with a return value, shouldn't the user be able to get that value from the script call?

scanner-darkly, 2017-09-20:
that'd be neat but i would prefer it to be more explicit, like setting `RETURN` variable or something (i would take more variables over this though)

tehn, 2017-09-20:
i like the way you're thinking, but this implementation i believe does not follow the explicit nature of the current tt syntax. it is prone to runtime error.

ie:

1. X ADD X SCRIPT 2
2. (blank)

since script 1 gets parsed at command entry, it has no way of knowing if script 2 will return a val or not. if script 2 becomes invalid, even just assuming a returned 0 to me is not an adequate behavior.

this is also a difficult concept to add to the TT knowledge base-- it is unlike the existing syntax (though your proposal makes sense to me) as the documentation isn't training to user to think about the stack. it'd be much better to just suggest using a temp variable for returning vars.

i know there's an argument against this due to script length concerns, again a push for the timeline to rectify this.

closed 2017-09-20 by tehn