monome/teletype #104: scenes with `#` in description don't load properly

scanner-darkly, 2017-09-24:
if a scene has a `#` character in the description when such scene is loaded everything after the `#` will not be read, including all of the scripts.

burnsauce, 2017-09-28:
Given that # is reserved in the teletype scene file syntax, I presume that the fix should be to disallow # in the scene description?

samdoshi, 2017-09-28:
Sounds like that's what we need to do until we rewrite the USB code.

It should be as simple as adding a branch to match `#` in [`process_preset_w_keys`](https://github.com/monome/teletype/blob/master/module/preset_w_mode.c#L37).

scanner-darkly, 2017-09-28:
perhaps a better option would be to allow entering it but replace it with something else at that point (a space?), not just ignore it - otherwise it might look like keyboard is not working, and there is no messaging provided to the user to clarify what's going on.

cmcavoy, 2017-10-01:
I looked for existing C serializer libraries we could leverage. I couldn't find any that seemed usable, but did see mention of using `/* */` for comments in this [implementation](http://www.happyponyland.net/cserialization/readme.html). Could we use an alternate comment format in TT scripts?

It's going to take me a while to understand the constraints around embedded development, but my first idea reading this thread was to leverage an existing serializer, doesn't sound like it's possible here - the embedded world is fun to learn.

samdoshi, 2017-10-01:
I think discussions here are being overtaken by discussions at:

https://llllllll.co/t/teletype-firmware-usb-disk-mode-interface/9442

We are currently contemplating switching from a text file format to a binary format.

jlmitch5, 2017-10-09:
> perhaps a better option would be to allow entering it but replace it with something else at that point (a space?), not just ignore it - otherwise it might look like keyboard is not working, and there is no messaging provided to the user to clarify what's going on.

My guess would be someone would be typing the `#` key because they are trying to document the script. i.e.:

```
#1:
foo

#2:
bar

#3:
baz
```

In that situation I don't really think replacing `#` with anything would really make sense to the user.  My gut is that just not responding to the character press is probably the least invasive action to take pre-changing all the big, serializer changes.

cmcavoy, 2017-10-10:
Can someone point to the serialization / deserialization code? Just curious to see it but have had trouble finding it in the source.

scanner-darkly, 2017-10-10:
https://github.com/monome/teletype/blob/master/module/usb_disk_mode.c

PR #306 (2023-04-10): Accept all line ending styles for USB load
#### What does this PR do?

previously, only *nix style line endings (LF) were accepted. i've updated deserialization code to accept DOS (CRLF) and Mac (CR) line ending styles as well. i also used it as an opportunity to make the deserialization code more robust.

as a consequence of this update, now it will also ignore any lines that start with `#` and are not followed by M, I, P, G or 1-8, which means script files can be annotated with comments. it also effectively fixes #104. 

while working on this, i realized that we use magic numbers for script  [... cut]
closed 2023-04-10 by scanner-darkly