monome/teletype #108: Tilde key does not open TRACKER

burnsauce, 2017-10-01:
The documentation states that the tilde (~) key will open tracker mode, but it doesn't.  Instead, it displays the £ character.

#### What steps are required to reproduce the bug?
1. Press the tilde key

samdoshi, 2017-10-01:
The `~` key used to do that, but I removed it in 2.0. It was replaced with the `<tab>` key cycling between the 3 modes, and the extra keys to jump directly to a mode.

Which docs still include the `~`?

cmcavoy, 2017-10-01:
I added a tag to this, not sure if the wording is right. `needs documentation` is supposed to mean "things that people could work on that are just documentation." If the tag isn't clear, feel free to change it.

burnsauce, 2017-10-01:
Reference is in docs/quickstart.md:142

commit 273864f7 (2017-10-09): Fixes #108 ~ key docs bug
commit baaec953 (2017-10-09): Merge pull request #113 from jlmitch5/patch-1

Fixes #108 ~ key docs bug
closed 2017-10-09 by tehn