monome/teletype #112: New Event Scheduler

burnsauce, 2017-10-02:
The AT (now R for repeat) feature requires a new scheduler to be implemented to prevent the inrush of equal-delay scripts from bogging down `tele_tick()`, causing i2c, trigger, slew, and display problems.

I propose a high-resolution priority scheduler to allow accurate synchronous events and fast response to asynchronous events.

Dewb, 2025-12-15:
Goals of a new scheduler could be
1. Prevent "heavy" scripts from interfering with input
2. Allow some restrictions to be relaxed, like the 10ms delay granularity.

In some ways these goals are opposed, but if overtaxing the hardware doesn't interfere with keyboard events etc., we can let users get themselves into trouble if they can get themselves out.  

Here's a very small/simple behavior that I think could be improved. DEL.X and DEL.R behave more differently than their documentation and purpose would suggest. `DEL.X 3 50: <command>` will schedule the command at 50, 100, and 150ms from now, while `DEL.R 3 50: <command>` will schedule delays at 1, 51, 101ms from now. With the 10ms delay processing granularity, this pushes events into the next delay-tick. (IIRC TR.TIME is 20 in this screenshot.)

[image]

tehn, 2025-12-15:
I'd support these changes!