monome/teletype #156: some keyboards losing keystrokes

jlmitch5, 2018-07-30:
[llllllll post](https://llllllll.co/t/mechanical-keyboard-for-teletype/5859/22?u=jlmitch5)

Present with Vortex Core (2.3 RC4/3.0 RC5), Vortex Pok3r (2.2/3.0 RC5).

Sometimes key strokes seem to be lost.  I (on my Vortex Core) can somewhat reliably reproduce by pressing three keys right after each other in super quick succession (ex. “A”, “D”, “H”), where I’ll only get “AD”.

scanner-darkly, 2018-07-30:
Vortex Core (2.3 RC4/3.0 RC5) - does it work on 2.2?

jlmitch5, 2018-07-30:
According to original poster, issue is still present on 2.2 (with Pok3r)

timcosgrove, 2018-07-30:
I posted about Vortrex Pok3r. I'm happy to test with any firmware, etc. I don't have a Monome development environment set up as yet but I can work on that to try out potential fixes.

timcosgrove, 2018-07-30:
I have https://github.com/scanner-darkly/teletype/tree/dev checked out, and am able to build from it. I am not a C developer, but if there's anything I can do to be useful re: testing, diagnostics etc, let me know.

scanner-darkly, 2018-07-30:
@timcosgrove thanks, i'll ping you on lines

PR #161 (2018-09-06): Dev3
#### What does this PR do?

Fixes #156 

#### Provide links to any related discussion on [lines](https://llllllll.co/).

https://llllllll.co/t/mechanical-keyboard-for-teletype/5859/22?u=jlmitch5

#### How should this be manually tested?

see the thread

#### I have,
* [x] updated CHANGELOG.md
* [x] updated the documentation
closed 2018-09-06 by scanner-darkly