monome/teletype #206: (meta) teletype issue forgiveness

tehn, 2020-07-03:
only 3 out of 20 of the issues are less than a year old.

many are stale feature requests and propositions--- many of initiated by myself--- which are largely no longer relevant as the system has shifted dramatically in some ways.

the norns crew recently did a massive purge of stale/won't-fix issues and it _felt amazing_.

i'm proposing something similar here, and since we're not co-located (and drinking beer), i think it'd be good to weigh in on specific issues prior to removal... though also perhaps there isn't much to discuss?

i'm going to use this issue as a thread to list possible issues to close. any comments are welcome. after some period or time i'll execute the agreed closures.

tehn, 2020-07-04:
reverse chrono

- #6 - obsolete, see https://github.com/monome/ii for discussion on ii
- #7 - stale
- #8 - it was a good idea, but huge feature adds don't fit the issue tracker. i can copy-paste the design idea to a wiki page titled "ideas!!!!" or something
- #9 - same as above, but i think maybe some features introduced actually allow for some of this? `R` i think?
- #12 - nope, weirdly specific
- #13 - just use a cable, it's eurorack
- #17 - stale
- #26 - i'm pretty sure that @scanner-darkly has tuned this?
- #42 - stale
- #74 - stale
- #81 - i thought the usb system was in a good place now?
- #104 - is this fixed??
- #107 - still weird?
- #112 - i think this makes sense (someone correct me?) and maybe is worth leaving in for the very-ambitious...
- #145 - @csboling did this in ansible, so there is precedent if someone wants to take it on in the future
- #155 - should stay open (?) but be renamed ie create key quick-reference
- #162 - fairly certain this has been implemented?
- #187 - i actually think the manual looks nice as is... happy to discuss
- #193 - ??
- #195 - likely still an issue

yes, i am proposing the closure of all but maybe 3-5 of the above. comments welcome!!

csboling, 2020-07-04:
> #7 - stale

There's been some user interest off and on related to alternate tracker UI but the main one I'm aware of is to have binary / hex / octal / base12 number visualization modes for working with bitwise ops and semitones rather than graphical representations.

> #9 - same as above, but i think maybe some features introduced actually allow for some of this? R i think?

You can trigger scripts programmatically and do enough interesting delay stuff that it covers a lot of use cases here I think.
The `R` op is not related to this but is instead a random number generator with configurable bounds via `R.MIN` / `R.MAX`.

> #81 - i thought the usb system was in a good place now?

There is significant work towards a simple USB disk menu system on this branch: https://github.com/burnsauce/teletype/tree/usb This might be interesting to revisit at some point but I think supporting the Timeline feature was one of the main motivators for reworking the USB disk code at the time this was done? Would probably need some other USB disk use cases to warrant a big push here.

> #104 - is this fixed??

Indeed this is still broken! Comments suggest that the USB disk mode might need rework for this, but may be possible to find a simpler resolution.

#107 - still weird?

Yes, the PDF section links are generally off by one page, but the HTML section links work fine.

> #145 - @csboling did this in ansible, so there is precedent if someone wants to take it on in the future

Yeah, seems like a good idea and should be straightforward.

> #162 - fairly certain this has been implemented?

Yes, this is `DEVICE.FLIP` since v3.1.0.

> #193 - ?? 

It appears that the documentation is incorrect and `LAST SCRIPT` always returns 0 in live mode. Would be a simple change to bring this op in line with the documentation.

> #195 - likely still an issue

Yeah, it has weird behavior on mobile that makes it difficult to browse.

---

The others mentioned I don't think I really have an opinion about.

scanner-darkly, 2020-07-04:
> #7 - stale

i do think this is a good idea, especially with more interest around trackers lately! and feels like this is a bigger feature, maybe move it to "ideas" as well?

> #9 - same as above, but i think maybe some features introduced actually allow for some of this? R i think?

there was some discussion [here](https://llllllll.co/t/teletype-3-2-feature-requests-and-discussions/32052/3), and something close can be achieved by using `DEL.R` but my understanding was that it's not 1:1 replacement due to limited iteration count.

> #26 - i'm pretty sure that @scanner-darkly has tuned this?

i believe this was fixed by @csboling 👍 [here](https://github.com/monome/teletype/pull/183)

> #42 - stale

this was actually addressed, you can select/copy/paste several lines now.

> #104 - is this fixed??
> Indeed this is still broken! Comments suggest that the USB disk mode might need rework for this, but may be possible to find a simpler resolution.

unless we change serialization completely this would probably require escaping `#` character, like replacing it with two `##` - the issue feels pretty edge casey to me though.. seeing as i'm the one who opened it i would be okay with closing it :)

> #112 - i think this makes sense (someone correct me?) and maybe is worth leaving in for the very-ambitious...
@burnsauce mentioned working on it, but yeah this is a super ambitious change.

> #187 - i actually think the manual looks nice as is... happy to discuss

i was going to post actually - using the HTML manual on ipad is slow - i think it's trying to render everything or something. i would say it's not very usable on ipad right now.

agreed on the rest!

tehn, 2020-07-16:
closed out agreed upon issues. we're down to a reasonable number now :)

closed 2020-07-16 by tehn