monome/teletype #216: v3.2.0 Script mute does not affect scripts triggered by $

cristianvogel, 2020-07-27:
Not sure if this is a bug or intentional? 

If I have a script that triggers more scripts using the `$` operator, the triggered script executes even if it is muted.

scanner-darkly, 2020-07-27:
it's intentional but i'm not sure what the reasoning was for muting triggers instead of muting scripts themselves. we should update the language in the manual to reflect that.

i don't see why we can't support both though. one way we could implement it without changing the live screen UI (since it currently shows trigger mutes only) is to have a shortcut to comment out all lines in a script (something like `CTRL-SHIFT-F..`).

attejensen, 2021-03-10:
How about a value of 2 mutes script triggers, 3 mutes both.

IOW binary
bit 0: mute jack triggers
bit 1: mute script triggers

00: nothing muted
01: jacks muted
10: scripts muted
11: both muted

Not sure how that should be shown on the UI, maybe:
bright = nothing muted
dimmed = one thing muted
blank= both muted

..Or if it's possible to blink, blinking could represent the 4th state.