monome/teletype #241: Problems installing avr32

cristianvogel, 2021-03-31:
#### Please describe the bug.
curl: (7) Failed to connect to distribute.atmel.no port 80: Network is unreachable
make: *** [downloads/avr32-patches.tar.gz] Error 7

#### What steps are required to reproduce the bug?
1. follow build instructions at https://github.com/monome/libavr32

cristianvogel, 2021-04-01:
I get this at 

`PREFIX=$HOME/avr32-tools make install-cross`

here is a log of commands based on instructions at `https://github.com/monome/libavr32`

```
Ramon-MBP:~ cristianvogel$ cd avr32-toolchain
Ramon-MBP:avr32-toolchain cristianvogel$ PREFIX=$HOME/avr32-tools make install-cross
cd downloads && curl -LO http://distribute.atmel.no/tools/opensource/Atmel-AVR32-GNU-Toolchain/3.4.2/avr32-patches.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:07 --:--:--     0
curl: (7) Failed to connect to distribute.atmel.no port 80: Network is unreachable
make: *** [downloads/avr32-patches.tar.gz] Error 7
Ramon-MBP:avr32-toolchain cristianvogel$ echo $HOME
/Users/cristianvogel```

scanner-darkly, 2021-04-01:
we really need to update those instructions..
are you building on linux? try this:

```
git clone https://github.com/monome/avr32-toolchain-linux.git deps/avr32-toolchain-linux
cd deps/avr32-toolchain-linux
tar xvfz avr32-gnu-toolchain-3.4.3.820-linux.any.x86_64.tar.gz
mv avr32-gnu-toolchain-linux_x86_64 $HOME/avr32-tools
unzip avr32-headers-6.2.0.742.zip -d $HOME/avr32-tools/avr32/include
```

cristianvogel, 2021-04-01:
Thanks! I’m trying to build on Mac OS actually 

🔝

scanner-darkly, 2021-04-02:
i was going to suggest trying using this: https://github.com/denravonska/avr32-toolchain
but according to somebody else who tried it, they couldn't get it to work on mac: https://llllllll.co/t/module-firmware-development-avr32-toolchain/7882/175

the option that did work for them was using a docker image.

cristianvogel, 2021-04-03:
The Docker image route is painless indeed.

closed 2021-04-03 by cristianvogel