monome/teletype #283: QT.B and QT.BX broken with negative input

attejensen, 2022-02-05:
#### Please describe the bug.
QT.B and QT.BX are broken with negative input

#### Please include the hash displayed at startup and the release of the software (e.g. "Teletype v2.0.1 5f838c9")
this is reproduced with the simulator in current (2022-02-05) git

#### What steps are required to reproduce the bug?
n.b 10 1189 // scale = Bb, C, Eb, F, Ab or N=10,12,15,17,20
vn n.b 1 // return 10 OK
vn n.b -1 // return 5 OK
vn qt.bx 0 n 0 // return 0 OK
vn qt.b n 1 // return 0 OK
vn qt.b n 2 // return 3 OK
vn qt.b n 4 // return 3 OK
vn qt.b n 5 // return 5 OK
vn qt.b n 6 // return 5 OK
vn qt.b n 7 // return 8 OK
vn qt.b n 8 //return 8 OK
vn qt.b n 9 //return 10 OK
vn qt.bx 0 n -1 // return 0 OK
vn qt.bx 0 n -2 // return -3 should be -2
vn qt.b n -2 // return -3 should be -2
vn qt.b n -5 // return -5 should be -4

desolationjones, 2022-02-05:
Negative voltage input gets improperly mirrored around zero. Working on the fix now.

desolationjones, 2022-12-31:
Fixed in PR #284

closed 2023-03-21 by scanner-darkly