monome/teletype #297: Time for a point release?

desolationjones, 2022-12-31:
There are a lot of users expressing desire for various betas to be merged and released. I think there are also some open issues and improvements we'd like to accomplish in the near term, but it seems like the balance may have recently tipped in favor of "release what we have". I'd like to open the floor for discussion 🙇 

Current "What's New":
- **FIX**: fix off-by-one error in `P.ROT` understanding of pattern length
- **FIX**: fix `CROW.Q3` calls `ii.self.query2` instead of `ii.self.query3`
- **FIX**: cache currently-running commands to avoid corruption during SCENE ops.
- **FIX**: delay when opening docs
- **FIX**: `PROB 100` would execute only 99.01% of the time.
- **FIX**: some `G.FDR` configurations caused incorrect rendering in grid visualizer
- **FIX**: fix `EX.LP` not returning correct values
- **FIX**: fix `QT.B` handling of negative voltage input
- **IMP**: scene load/save code refactor, add scene load/save tests
- **IMP**: fader ops now support up to four faderbanks
- **NEW**: new Disting EX ops: dual algorithms, `EX.M.N#`, `EX.M.NO#`, `EX.M.CC#`
- **FIX**: reset M timer when changing metro rate

Open issues - documentation:
- TT Queue documentation gives 2 different values for maximum Q size
- N.CS lookup tables and documentation/help errors
- TinyTeX as MacTeX alternative
- Teletype web manual at mobile screen widths needs menu fix
- docs: utilize monome/docs css
- PDF links in table of contents are not quite right.

Open issues - operations/enhancement:
- TT: copying Queue to a pattern erases complete pattern
- Pattern start/end/length/indexing needs an overhaul
- v3.2.0 Script mute does not affect scripts triggered by $
- CV output calibration
- New Event Scheduler enhancement
- tracker alternate data views enhancement

Open issues - bugs:
- The STATE command does not take into account the status of DEVICE.FLIP
- TR.TIME shorter than 11ms gets skipped by delay MODs
- scenes with # in description don't load properly
- MI.NCH op seems to return 0-15 instead of 1-16

Betas, forks, and improvements under development:
- i2m2midi OPs (@scanner-darkly)
- Disting Ex dual algos and MIDI enhancements (@scanner-darkly)
- Calibration OPs (@Dewb)
- USB file management (@beels, @burnsauce)
- Dual W/2 (@desolationjones) 
- Arc euclidean OPs (DocPolyester)
- Sinfonian via i2c2midi hack (subvertnormality)
- WORDs and stack manipulation (trentgill)
- Various "script count extension" hacks

scanner-darkly, 2022-12-31:
Disting Ex dual algos PR has been merged already.

also there is an open PR: https://github.com/monome/teletype/pull/277

tehn, 2023-01-02:
i'm up for a point release

Dewb, 2023-01-03:
The calibration op has been sitting mostly done for a while, just need to finish rewriting it using integer math and decide exactly where the calibration data should be stored. It's probably doable to get it in this month, but due to the nature of the change it would be good to have an extended hardware beta testing period, so I wouldn't hold the release for it.

Dewb, 2023-01-03:
re: "delay when opening docs", did #276 ever get deployed?

scanner-darkly, 2023-01-04:
looks like the official manual only includes what's in the latest official version 4.0. once we're ready to make a release we can update the docs as well.

i would love for the following to be in the next release, let me try and work on it in the next week/two if that's okay:
- i2m2midi OPs
- scenes with # in description or ops don't load properly

i might also tackle this if it turns out to be an easy fix:
- The STATE command does not take into account the status of DEVICE.FLIP