monome/teletype #337: First PN.RND after P.SEED isn't random

mikeyford, 2024-11-18:
#### Please describe the bug.

The first call to `PN.RND` always returns the same value, after seeding the randomization via `P.SEED`. Subsequent calls of `PN.RND` are random.

I'm happy to look into this, but please let me if you have any pointers or I'm misunderstanding the feature.

#### Please include the hash displayed at startup and the release of the software (e.g. "Teletype v2.0.1 5f838c9")

Teletype v5.0.0 00FDFD2

#### What steps are required to reproduce the bug?
fill pattern 0 with values
`P.SEED x`
`PN.RND 0`
will always return the same value, regardless of `x`

Alternatively,  filling 0 with values, and pattern 1 with seeds, then create a script like:
```
P.SEED PN.NEXT1
PN.NEXT 2 PN.RND 0
```
then trigger the script repeatedly to fill pattern 2 with the same value

Dewb, 2024-11-18:
I can reproduce this. The behavior is consistent with a call to random_next() with an uninitialized random state (r and z both 0), but from a quick skim of the code it looks like the pattern random state should be initialized alongside all the other random data in the scene state, none of which exhibit this issue. Could be possible that something is accidentally stomping on just the pattern random state after initialization.

mikeyford, 2024-11-19:
I think because we're using a constant to initialize `z` [here](https://github.com/monome/libavr32/blob/eb834b98742219d2b5a08e523fc0e5d9000e97ec/src/random.c#L3C1-L6C2).  

I don't have a C toolchain here at work, but I was able to reproduce with the algorithm in Python. The numbers produced _are_ different, but are similar, especially at the end. We're taking the mod of the random value to choose from the pattern, which is I guess this is why it hasn't been so obvious in other contexts?

Switching to `r->z = ~seed`

Will test properly with Teletype when I'm home and throw up a PR in the next day or two.