monome/teletype #342: Proposal: Centralize definition of OPs into a single file

Dewb, 2025-07-06:
Users must read or touch many files to reason about or define a new OP. Currently that list includes the tele_ops table, the Ragel parser, the module, test, and simulator Makefiles, the help mode documentation, and the paper documentation.

We're already doing a little bit of codegen to create op_enums.h, but lots of this other material could be generated also.

The ideal would be a single file that describes all the information for an OP or small collection of related OPs, so adding or altering behavior could target a single location in the source tree. If a single file was untenable, then two separate files for definition and implementation in the same folder would also be an improvement.

In addition to making adding OPs easier, this change would also make it easier to refactor the way OPs are processed and documented, because all the implementations could be changed by changing the codegen.

tehn, 2025-07-08:
This would be a huge improvement! But in interest in preserving your efforts, is the OP adding checklist in place already? If not maybe some contrib doc addition would be sufficient.

Dewb, 2025-07-09:
(I am still open to counterargunents and might talk myself out of this, this is just a place to gather thoughts! I’m leaning towards two files to define a set of ops, a c file and some kind of structured text.)